chainsx/ubuntu64-rpi #25: Errors were encountered while processing:  phddns:armhf

ourfor, 2018-07-15:
(Reading database ... 24637 files and directories currently installed.)
Preparing to unpack oray.deb ...
Remove Phddns Service Success.
Unpacking phddns:armhf (3.0.2) over (3.0.2) ...
Setting up phddns:armhf (3.0.2) ...
/usr/orayapp/oraysl: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
dpkg: error processing package phddns:armhf (--install):
 installed phddns:armhf package post-installation script subprocess returned error exit status 127
Processing triggers for systemd (237-3ubuntu10) ...
Errors were encountered while processing:
 phddns:armhf
安装花生壳官网的树莓派安装包遇到的，尝试过安装包含libstdc++.so.6的包，但问题依旧存在

chainsx, 2018-07-16:
需要开启armhf的支持
https://github.com/chainsx/ubuntu64-rpi/wiki/32-Bit(armhf)-Support

ourfor, 2018-07-16:
就是开启这个遇到的

ourfor, 2018-07-16:
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic InRelease
Get:2 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:3 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:4 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-security InRelease [83.2 kB]
Get:5 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-updates/main armhf Packages [156 kB]
Get:6 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-updates/main arm64 Packages [168 kB]
Fetched 571 kB in 3s (184 kB/s)
Reading package lists... Done
 ⚡ root@Sagit  ~  apt-get install libc6:armhf
Reading package lists... Done
Building dependency tree
Reading state information... Done
libc6:armhf is already the newest version (2.27-3ubuntu1).
The following packages were automatically installed and are no longer required:
  libasan5-arm64-cross libgcc-8-dev-arm64-cross liblsan0-arm64-cross
  libstdc++-8-dev-arm64-cross libtsan0-arm64-cross libubsan1-arm64-cross
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up phddns:armhf (3.0.2) ...
/usr/orayapp/oraysl: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
dpkg: error processing package phddns:armhf (--configure):
 installed phddns:armhf package post-installation script subprocess returned error exit status 127
Errors were encountered while processing:
 phddns:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)

railjty, 2018-07-19:
这个有办法，sudo apt-get install libstdc++6:armhf -y

closed 2018-08-03 by chainsx