chainsx/ubuntu64-rpi #28: gnome循环登陆无法进入系统

KevinJia2011, 2018-08-30:
你好，自从购买3b+之后一直在跟进关注你这个项目。
在按照教程终于安装好了gnome之后，无法进入系统，希望下次修复，谢谢！
xinit: server error
xauth: timeout in locking authority file /home/ubuntu/.Xauthority

chainsx, 2018-08-31:
如果电脑启动后显示用户登录界面，输入用户名和密码但进不到桌面，一直在登陆界面循环。
进入终端后使用startx命令，看是否有错误提示，如果是timeout in locking authority file in /home/**/.Xauthority,解决办法是将.Xauthority删掉，在检查用户目录权限，确定为775，再reboot重启就OK了。

chainsx, 2018-08-31:
这个不是系统问题

whitwolf, 2018-09-03:
持续关注

KevinJia2011, 2018-09-05:
我按ctrl+alt+F1 没有任何反应，无法进入终端； tty2~6可以进入但是按照操作之后没有任何作用

mason086, 2018-09-19:
我下载下来使用imageusb烧写到tf卡中，启动，并在控制台安装lightdm；也是输入密码后登录不正确，我在raspi-config的boot option中设置启动到登录界面。   为了减少差异，我默认的使用账户：ubuntu，密码：ubuntu； 我在/home/ubuntu下并没有看到 ./Xaut...文件（ls -all查看）；  然后我就安装了gnome，也是一样的。 我发现我在控制台中输入`sudo startx `，提示命令不存在； 所以我继续安装xinit
（`sudo apt-get install xinit`）  之后我就看到了 ./Xaut... 文件。 我安装chainsx说的方式，删除这个文件或者修改目录权限都没成功。 
后来找到：
https://blog.csdn.net/LBO4031/article/details/8509495
和
https://blog.csdn.net/maxilbert/article/details/49754619
怀疑就是目录权限的问题。 我用ls -all查看，确实，权限是root的； 所以，我使用
`sudo chown -R ubuntu:ubuntu /home/ubuntu`
并再次使用`ls -all `查看，确定权限变为用户ubuntu，再次重启，就ok了！

mason086, 2018-09-19:
我会在[这里](http://note.youdao.com/noteshare?id=e568b90f990c0817e3c678486c03813d&sub=B16C6D7300664083926A3290A933491E)分享一些试用过程大家。

Quzard, 2019-05-13:
我按照操作之后没有任何作用