chainsx/ubuntu64-rpi #31: 第一次 boot 就错误？

CodeNemo, 2018-09-27:
串口通信连接看到的信息，网络接口指示灯不亮，系统未启动。

[    0.102653] dmi: Firmware registration failed.
[    0.707020] Error: Driver 'sdhost-bcm2835' is already registered, aborting...
[    0.919811] EXT4-fs (mmcblk0p2): error loading journal
[    0.925988] Kernel panic - not syncing: VFS: Unable to mount root fs on unkno            wn-block(179,2)
[    0.929771] Indeed it is in host mode hprt0 = 00021501
[    0.939993] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 4.14.66-chainsx-cxcore+             #2
[    0.941631] NOHZ: local_softirq_pending 80
[    0.949630] NOHZ: local_softirq_pending 80
[    0.953630] NOHZ: local_softirq_pending 80
[    0.957630] NOHZ: local_softirq_pending 80
[    0.961630] NOHZ: local_softirq_pending 80
[    0.965630] NOHZ: local_softirq_pending 80
[    0.969630] NOHZ: local_softirq_pending 80
[    0.977130] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[    0.983144] Call trace:
[    0.985683] [<ffffff85d3c8b1e0>] dump_backtrace+0x0/0x270
[    0.991256] [<ffffff85d3c8b474>] show_stack+0x24/0x30
[    0.996473] [<ffffff85d43412cc>] dump_stack+0xac/0xe8
[    1.001688] [<ffffff85d3ca20ac>] panic+0x120/0x274
[    1.006637] [<ffffff85d45d1484>] mount_block_root+0x20c/0x260
[    1.012566] [<ffffff85d45d1728>] mount_root+0x124/0x13c
[    1.017960] [<ffffff85d45d18b8>] prepare_namespace+0x178/0x1c0
[    1.023978] [<ffffff85d45d0f4c>] kernel_init_freeable+0x220/0x240
[    1.030266] [<ffffff85d4352c08>] kernel_init+0x18/0x110
[    1.035658] [<ffffff85d3c85250>] ret_from_fork+0x10/0x18
[    1.041144] SMP: stopping secondary CPUs
[    1.045215] Kernel Offset: 0x5cbc00000 from 0xffffff8008000000
[    1.051229] CPU features: 0x0802004
[    1.054826] Memory Limit: none
[    1.057993] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs             on unknown-block(179,2)

CodeNemo, 2018-09-27:
解压 .zip 后，用的是 rufus 写入镜像。

chainsx, 2018-10-02:
没有叫你解压zip啊。。。。。
哪来的zip。。。。。

closed 2018-10-02 by chainsx