chainsx/ubuntu64-rpi #42: 关于声卡问题

sives1, 2018-12-10:
直接:
aplay -l
提示:
aplay:device list:270:no soundcards found...
所以是不支持声卡哇⊙ω⊙?
无敌SCP-CN-666-1前来问候

sives1, 2018-12-11:
成功解决!!!
首先改/boot/config.txt
<code>sudo nano /boot/config.txt</code>
将
<code>dtparam=i2c_arm=on,audio=on0</code>
替换成
<code>dtparam=i2c_arm=on</code>
<code>dtparam=audio=on</code>
然后终端输入
<code>sudo amixer cset numid=3 1</code>
好了,享受有声音的日子吧~
        无敌地球喵喵教
初三党

sives1, 2018-12-11:
警告:板载声卡和USB声卡和以上设置
完全一样!!!(才怪)
需要执行:
<code>amixer contents</code>
找到你要的声卡的numid
然后执行:
<code>sudo amixer cset numid=\<numid\></code>
比如假设我的USB声卡numid为6
那么执行:
<code>sudo amixer cset numid=6</code>
(其实你可以直接:<code>sudo amixer cset numid=1 2 3 4 5 6 7 8 9</code>/手动滑稽)

buzai, 2019-04-25:
66666