chainsx/ubuntu64-rpi #51: 可以提供内核header吗

leonunix, 2019-05-21:
想在上面编译一个wireguard使用
可是提示
make[1]: *** /lib/modules/4.14.66-chainsx-cxcore+/build: No such file or directory.  Stop
找遍github的repo都找不到内核header在哪里
忘可以提供

chainsx, 2019-06-12:
https://github.com/chainsx/firmware64-rpi/tree/master/install/usr/include/linux

chainsx, 2019-06-12:
https://github.com/chainsx/firmware64-rpi/tree/master/install/usr/include

chainsx, 2019-06-12:
在编译系统时已经加入了内核头文件的，在编译应用程序的时候，在GCC的CFLAG参数中添加“-I(内核源码路径)/usr/include”，这样编译器就知道在编译时找到相关的头文件。

chainsx, 2019-07-05:
已经修复，
https://github.com/chainsx/firmware64-rpi/releases

leonunix, 2019-07-05:
测试了新的内核。安装过程产生了失败。貌似您少提供了一个包
Selecting previously unselected package linux-image-5.2.0-rc7-cxcore-chainsx-devel+.
(Reading database ... 65532 files and directories currently installed.)
Preparing to unpack linux-image-5.2.0-rc7-cxcore-chainsx-devel+_5.2.0-rc7_arm64.deb ...
Done.
Unpacking linux-image-5.2.0-rc7-cxcore-chainsx-devel+ (5.2.0-rc7-cxcore-chainsx-devel+-10.00.Custom) ...
Setting up linux-image-5.2.0-rc7-cxcore-chainsx-devel+ (5.2.0-rc7-cxcore-chainsx-devel+-10.00.Custom) ...

 Hmm. There is a symbolic link /lib/modules/5.2.0-rc7-cxcore-chainsx-devel+/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/5.2.0-rc7-cxcore-chainsx-devel+/build

 Hmm. The package shipped with a symbolic link /lib/modules/5.2.0-rc7-cxcore-chainsx-devel+/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/5.2.0-rc7-cxcore-chainsx-devel+/source

Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.2.0-rc7-cxcore-chainsx-devel+ /boot/vmlinuz-5.2.0-rc7-cxcore-chainsx-devel+
run-parts: executing /etc/kernel/postinst.d/dkms 5.2.0-rc7-cxcore-chainsx-devel+ /boot/vmlinuz-5.2.0-rc7-cxcore-chainsx-devel+
Error! Your kernel headers for kernel 5.2.0-rc7-cxcore-chainsx-devel+ cannot be found.
Please install the linux-headers-5.2.0-rc7-cxcore-chainsx-devel+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 5.2.0-rc7-cxcore-chainsx-devel+ /boot/vmlinuz-5.2.0-rc7-cxcore-chainsx-devel+